OPCFoundation/UA-AnsiC-Legacy #26: [Armv6]: Compilation on different architectures

miraenator, 2016-12-12:
Hi, 

I tried to compile the master branch on Raspberry Pi. However, the default fallback (`linux_gcc_rules.mak`) is to i386 architecture. The rules only consider `x86_64` and other (=`i386`).

Therefore the code does not compile on Raspberry Pi.

Also, the `uname -i` on Rpi returns the value `unknown`. The command `uname --machine` can be used to obtain more information.

Solution (suggest):
1) Be ready for different architectures so that a new architecture can be added easily. Not sure if the `i386` fallback is a good idea. Maybe you should at least print an error/warning message that you are falling back to this architecture.
2) In case the `--hardware-platform` is `unknown`, get more information from `--machine` parameter of the `uname` command.

```
$ git diff .
diff --git a/linux_gcc_rules.mak b/linux_gcc_rules.mak                                                                                      
index 6e9ef8b..f37e7d4 100644                                                                                                               
--- a/linux_gcc_rules.mak                                                                                                                   
+++ b/linux_gcc_rules.mak                                                                                                                   
@@ -23,10 +23,15 @@ MKDIR = mkdir -p
 
 ifndef MACHINE_TYPE
 ifndef CROSS_COMPILE
-       TMP_MACHINE_TYPE = $(shell uname -i)
+       TMP_MACHINE_TYPE = $(shell uname --hardware-platform)
+       ifeq ($(TMP_MACHINE_TYPE), unknown)
+               TMP_MACHINE_TYPE = $(shell uname --machine)
+       endif
        # sometimes "unknown" is returned - default to i386
        ifeq ($(TMP_MACHINE_TYPE),x86_64)
                MACHINE_TYPE = x86_64
+       else ifeq ($(TMP_MACHINE_TYPE), armv6l)
+               MACHINE_TYPE = armv6
        else
                MACHINE_TYPE = i386
        endif
@@ -39,10 +44,10 @@ ifndef MACHINE_OPT
 ifndef CROSS_COMPILE
        ifeq ($(MACHINE_TYPE),x86_64)
                MACHINE_OPT = -m64
-       else
-       ifeq ($(MACHINE_TYPE),i386)
+       else ifeq ($(MACHINE_TYPE),i386)
                MACHINE_OPT = -m32
-       endif
+       else ifeq ($(MACHINE_TYPE),armv6)
+               MACHINE_OPT = -march=armv6
        endif
 endif
 endif
```

bernd-edlinger, 2016-12-12:
Hi,

for cross-compilation please use the following command:

./build-linux CROSS_COMPILE=arm-whatever-linux- MACHINE_TYPE=machine MACHINE_OPT="-march=armv6 -I <path-to-cross-openssl>"

if you are builting directly on your machine, just set MACHINE_TYPE and MACHINE_OPT

miraenator, 2016-12-12:
Ok, this would definitely work. 

Thanks.

**IMHO**
The Readme (and build_linux.sh script name) is therefore quite confusing: Newbies would not see the difference between 'linux for x86' and 'linux'. Also, the 'MACHINE_TYPE' and the 'MACHINE_OPT' are quite advanced for them. And provided that Raspberry Pi is a common platform, they might get confused if the 'build_linux.sh' script fails.

I leave the resolution/clarification to you. There are many ways. IMHO the most important (and this was my outline of the solution) is the new user experience -- so that the 'build_linux.sh' simply proceeds and builds.  

IMHO you might consider the following:
 - Clarify the compilation on other architectures (w. example) in Readme.md
 - Rename the build_linux.sh to ./build_linux_x86.sh
 - Improve the detection of the architectures (and print warning for the 'unknown', maybe with a hint how to proceed)
 - Adding an example compilation for Raspberry Pi (should be tested, I have only B+ model) 

**end IMHO**

bernd-edlinger, 2016-12-12:
Yes, thanks for your suggestions.

I will let this issue open as a reminder to add this topic in the readme.

bernd-edlinger, 2017-01-24:
We improved the build script, and added an example how to do a cross compilation.
Thanks for reporting this.

closed 2017-01-24 by bernd-edlinger