OPCFoundation/UA-AnsiC-Legacy #30: UA Stack does not compile with OpenSSL 1.1.0

bernd-edlinger, 2017-01-09:
OpenSSL 1.0.1 is no longer supported, and should not be used any more.
Everything works with OpenSSL 1.0.2, which is supported until end of 2019.

However the UA Stack does not compile with OpenSSL 1.1.0 currently,
because may objects are opaque in 1.1.0 and need access functions now,
BUT not all data members that we use are accessible at the moment.
So this needs to be discussed with the OpenSSL project.

cacamille3, 2017-01-19:
I have actually ported the Stack to MAC OS and used OpenSSL 1.1.0
There is indeed some few changes.

For example in the function OpcUa_P_OpenSSL_X509_GetSignature() accessing the signature of a X509 pointer is not allowed anymore, so you can not get the length of the certificate's signature.

``
       a_pSignature->Signature.Length = pX509Certificate->signature->length;
``

could be replaced with 

```
const X509_ALGOR *sig_alg;
const ASN1_BIT_STRING *signature;
X509_get0_signature(&signature, &sig_alg, pX509Certificate);
if(signature == OpcUa_Null)
{
    X509_free(pX509Certificate);
    uStatus = OpcUa_Bad;
    OpcUa_GotoErrorIfBad(uStatus);
}
a_pSignature->Signature.Length = ASN1_STRING_length(signature);
```

Will you be interested on a Pull request ?

bernd-edlinger, 2017-01-20:
The problem lies deeper than that.

Unfortunately I can not simply break compatibility to the old OpenSSL API,
as this would for instance not work in still maintained linux distributions.
So all these places need to be conditionalized.

Also in the moment I see no way how to access the chain from the X509_STORE_CTX
here:

```
static int OpcUa_SslSocket_VerifyCertificate( X509_STORE_CTX *ctx, void *arg)
{
    OpcUa_InternalSslSocket* pInternalSocket   = (OpcUa_InternalSslSocket*)arg;
    STACK_OF(X509)*          pChain            = ctx->untrusted;
```

This will need a new access function in the OpenSSL IMHO.

cacamille3, 2017-01-20:
There is one function in the 1.1.0 version of OpenSSL to do that as well

`STACK_OF(X509) *pChain = X509_STORE_CTX_get0_untrusted(ctx);`

Described as following in OpenSSL doc: ([source](https://www.openssl.org/docs/man1.1.0/crypto/X509_STORE_CTX_get0_untrusted.html)) 
> X509_STORE_CTX_get0_untrusted() retrieves an internal pointer to the stack of untrusted certificates associated with ctx.

cacamille3, 2017-03-20:
What about creating a new branch to fix compilation with OpenSSL 1.1.X ?

bernd-edlinger, 2017-03-20:
Support for OpenSSL 1.1 is on our roadmap for end of this year.

bernd-edlinger, 2017-06-12:
Implemented backward-compatible support for openssl 1.1.0 with 5422a7e.

closed 2017-06-12 by bernd-edlinger