OPCFoundation/UA-AnsiC-Legacy #31: Variant encoder/decoder: accepts encoding Null built-in id values (unspecified)

vmo6trl, 2017-01-13:
Hello,

I noticed the stack accepts to encode/decode Variant with a Null built-in Id value whereas it is not specified as valid in the OpcUa specification 1.03 (part 6, table 14). Moreover it is accepted only in case of Variant single value (in this case only encoding byte is encoded and then no value) and not in case of array/matrix value (bad status returned in those case).
In Stack/stackcore/opcua_binarydecoder.c line 2911 (commit 9e5d1b319e14f7c21f04ac16c42a8c1cd18fe6ac):
>switch(a_pValue->Datatype)
>        {
>            case OpcUaType_Null:
>            {
>                break;
>}

In Stack/stackcore/opcua_binaryencoder.c line 3539 (commit b5e5d6ab0f7297c84e6f950699ab519349d201ae):
>switch(a_pValue->Datatype)
>        {
>            case OpcUaType_Null:
>            {
>                break;
>}

Since it is not specified, it should be considered as invalid.

bernd-edlinger, 2017-01-20:
The Spec is wrong.

In case of DataValue empty Variants will not be needed, because it
encodes empty variants in one bit of the encoding byte.

But other more advanced data structures, like
OpcUa_EventFieldList may need to encode empty values.

SparhawkSoftware, 2017-01-23:
We will publish an errata for 1.03 that adds a statement to 5.2.2.16:
A value of 0 for the EncodingMask indicates that no Value is encoded. This is the representation of the NULL value for the Variant,

bernd-edlinger, 2017-01-23:
Thanks for reporting this.

closed 2017-01-23 by bernd-edlinger

vmo6trl, 2017-01-24:
As a consequence I added a note to indicate the errata reference in the specification issue I also opened:
https://opcfoundation-onlineapplications.org/mantis/view.php?id=3682
Regards