OPCFoundation/UA-AnsiC-Legacy #38: Always get segment fault after running for a while

zhengyuli, 2017-02-09:
I write a simple OPC UA client based on UA-AnsiC to read node values, the security mode and policy is None. After running for a while, the client always exit with segment fault.

The GDB traceback:
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffeeffd700 (LWP 94602)]
__memcpy_ssse3_back () at ../sysdeps/x86_64/multiarch/memcpy-ssse3-back.S:2109
2109		mov	-5(%rsi), %edx
#0  __memcpy_ssse3_back () at ../sysdeps/x86_64/multiarch/memcpy-ssse3-back.S:2109
#1  0x000000000080fecf in OpcUa_P_Memory_MemCpy (pBuffer=0x7fffe0049500, nSizeInBytes=5, pSource=0x3000300020002, nCount=5) at platforms/linux/opcua_p_memory.c:68
#2  0x000000000080e5a4 in OpcUa_Memory_MemCpy (a_pBuffer=0x7fffe0049500, a_nSizeInBytes=5, a_pSource=0x3000300020002, a_nCount=5) at core/opcua_memory.c:67
#3  0x0000000000823030 in OpcUa_TcpSecureChannel_Renew (a_pSecureChannel=0x7fffe0001070, a_hTransportConnection=0x111f8b0, a_ChannelSecurityToken=...,
    a_eMessageSecurityMode=OpcUa_MessageSecurityMode_None, a_pbsClientCertificate=0xc4203fcc60, a_pbsServerCertificate=0x7fffeeffc960, a_pNewReceivingKeyset=0x7fffe00493d0,
    a_pNewSendingKeyset=0x7fffe0000fe0, a_pNewCryptoProvider=0x7fffe0001270) at securechannel/opcua_tcpsecurechannel.c:571
#4  0x00000000008218e6 in OpcUa_SecureConnection_ProcessOpenSecureChannelResponse (a_pConnection=0x111f8b0, a_pTransportIstrm=0x7fffe0001380, a_uResponseStatus=0,
    a_bResponseComplete=1 '\001') at securechannel/opcua_secureconnection.c:3168
#5  0x000000000081f6c2 in OpcUa_SecureConnection_InternalOnResponse (a_pSecureConnectionInterface=0x111f8b0, a_pCallbackData=0x0, a_uResponseStatus=0,
    a_ppTransportIstrm=0x7fffeeffcb00, a_bResponseComplete=1 '\001') at securechannel/opcua_secureconnection.c:1814
#6  0x000000000081f370 in OpcUa_SecureConnection_OnNotify (a_pTransportConnection=0x111fca0, a_pCallbackData=0x111f8b0, a_eEvent=OpcUa_ConnectionEvent_Response,
    a_ppTransportInputStream=0x7fffeeffcb00, a_uConnectionStatus=0) at securechannel/opcua_secureconnection.c:1654
#7  0x0000000000886211 in OpcUa_TcpConnection_ProcessResponse (a_pConnection=0x111fca0, a_pInputStream=0x7fffe0001380) at transport/tcp/opcua_tcpconnection.c:534
#8  0x00000000008868ad in OpcUa_TcpConnection_ReadEventHandler (a_pConnection=0x111fca0, a_pSocket=0x111f9e8) at transport/tcp/opcua_tcpconnection.c:826
#9  0x0000000000886ce8 in OpcUa_TcpConnection_SocketCallback (a_pSocket=0x111f9e8, a_SocketEvent=1, a_pUserData=0x111fca0, a_nPortNumber=47896, a_bIsSSL=0 '\000')
    at transport/tcp/opcua_tcpconnection.c:1085
#10 0x00000000008a7288 in OpcUa_Socket_HandleEvent (a_pSocket=0x111f9e8, a_uEvent=1) at platforms/linux/opcua_p_socket_internal.c:914
#11 0x00000000008a7f75 in OpcUa_P_Socket_HandleFdSet (a_pSocketManager=0x111f670, a_pSocketArray=0x7fffeeffce00, a_uEvent=1) at platforms/linux/opcua_p_socket_internal.c:1352
#12 0x00000000008a78e4 in OpcUa_P_SocketManager_ServeLoopInternal (a_pSocketManager=0x111f670, a_msecTimeout=1000, bRunOnce=0 '\000')
    at platforms/linux/opcua_p_socket_internal.c:1139
#13 0x000000000089acb4 in OpcUa_P_SocketManager_ServerLoopThread (a_pArgument=0x111f670) at platforms/linux/opcua_p_socket_interface.c:59
#14 0x000000000089dc5c in pthread_start (args=0x111ff70) at platforms/linux/opcua_p_thread.c:62
#15 0x00007ffff79c4dc5 in start_thread (arg=0x7fffeeffd700) at pthread_create.c:308
#16 0x00007ffff76f373d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:113

bernd-edlinger, 2017-02-10:
You should debug your application with address sanitizer:

make -f linux_gcc.mak MACHINE_OPT="-g -fsanitize=address -fno-omit-frame-pointer"

zhengyuli, 2017-02-11:
The same error after adding these options

zhengyuli, 2017-02-14:
I tried official release 1.02 and didn't find this problem, this may be a bug in current master branch.

bernd-edlinger, 2017-02-14:
The call stack indicates that the stack crashes while accessing the client certificate.
Also for security mode none, there must be a dummy client certificate.  Make it a static for instance.

zhengyuli, 2017-02-14:
Hi @bernd-edlinger, thanks for your quick reply, let me have a try.

zhengyuli, 2017-02-15:
Hi @bernd-edlinger, I tried this method but still got the same error. From OpcUa_Memory_MemCpy (a_pBuffer=0x7fffd4049df0, a_nSizeInBytes=2, a_pSource=0x3, a_nCount=2) we can find that the length of client certificate is 2 but actually the length of client certificate passed to OpcUa_Channel_Connect is greater than 2. I think this certificate is not same as the certificate passed in OpcUa_Channel_Connect and most likely a certificate variable which has not been initialized. BTW, are there any other client examples that I can refer to.

zhengyuli, 2017-02-21:
Hi @bernd-edlinger, the same problem for SecurityMode=Sign&Encrypt, BTW, what's the purpose of OpcUa_P_SocketManager_ServerLoopThread.

closed 2017-03-01 by zhengyuli