OPCFoundation/UA-AnsiC-Legacy #41: opcua_endpoint_ex.c is missing

eckes, 2017-03-20:
In version 1.02-336.1, the file `stack/proxystub/serverstub/opcua_endpoint_ex.c` was contained in the sources. Browsing the history of this repo shows me that the file is missing from the beginning.

The rationale for that file was to be able to provide an own de-/encoder which improves performance and saves memory because you could directly deserialize into or serialize from your native data types.

We've used that functionality and now cannot update to 1.03.340 because of that file missing. At least not without putting a huge effort in re-creating all the conversion stuff that we don't need when using `opcua_endpoint_ex.c`.

Please reinclude that file in the distribution.

bernd-edlinger, 2017-03-20:
We have no test coverage whatsoever on this file, therefore it was discontinued at the time
when the https protocol was introduced, sorry.
But I do not understand why you can't directly use the SecureListener API.

eckes, 2017-03-22:
The SecureListener API is a good point. Thanks for the hint.

closed 2017-03-22 by eckes