OPCFoundation/UA-AnsiC-Legacy #43: OpcUa_Channel can't be deleted from BeginConnect's callback

alexsmn, 2017-03-26:
I'm implementing async client with OpcUa_Channel_BeginConnect. What I need is to cleanup associated state and channel itself when connection fails.

The problem is that when OpcUa_Channel_Delete is called from callback, it crashes, because channel's mutex being deleted is aquired by OpcUa_Channel_OnNotify.

Here is what I need (using C++ for simplicity):

struct State {
  OpcUa_Channel channel;
  // more members
};

OpcUa_StatusCode StateCallback(..., OpcUa_Void* a_pCallbackData, OpcUa_Channel_Event eEvent, ...) {
  State* state = (State*)a_pCallbackData;
  if (eEvent == eOpcUa_Channel_Event_Disconnected) {
    // Here I need to cleanup the state.
    OpcUa_Channel_Delete(&state->channel); // <-- crash here
    delete state;
    return;
  }
}

void Connect() {
  State* state = new State;
  OpcUa_Channel_BeginConnect(..., StateCallback, state);
}

I believe OpcUa_Channel_OnNotify needs to avoid calling the callback under locked mutex. Or there are more correct solutions?

Note that OpcUa_Channel_Connect uses wait and locks the channel mutex. Generally it's need only for |OpcUa_IsGood(uStatus)| cases, but it helps to avoid the problem, because it guarantees the the mutex is unlocked by the callback thread before return. And it works since callback thead always differs from the thread calling OpcUa_Channel_Connect.

bernd-edlinger, 2017-07-19:
No, you cannot self-destruct the channel from a callback, it will always dead-lock.
You should do all connect and disconnect calls from your own thread.

alexsmn, 2017-07-20:
Isn't it an artificial limitation? E.g. Boost.Asio socket can be perfectly destroyed from async callback.
Notifying under locked mutex is a way for dead-locks, unless all notifications are queued through separate thread. That in order brings user code design complication and performance inoptimality for queueing notification to separate thread.

bernd-edlinger, 2017-07-21:
Even if no mutex is locked, OpcUa_Channel_Delete will wait for the select thread to shut down,
which will be blocked in a callback that does not return.

alexsmn, 2017-07-21:
OK. I missed that each Channel owns a Thread used for socket selects and notifications. Or more precisely, each Channel owns a TcpConnection that owns a SocketManager that owns a Thread. While I expected single SocketManager backed by single ThreadPool.
Thank you for your answers.

closed 2017-09-01 by alexsmn