OPCFoundation/UA-AnsiC-Legacy #45: AnsiCSampleServer "CreateSubscription not Implemented"

joerg555, 2017-05-09:
I have compile the Stack and AnsiC Sample Server.
When i watch any variable in UaExpert, i get an CreateSubscription not Implemented.
Could you extend the example or show a way how to implement this.
i think the Entry point is:
```c++
OpcUa_ServiceType  dummy_CreatSubscription =
{
    OpcUaId_CreateSubscriptionRequest,
    OpcUa_Null, /*&OpcUa_CreateSubscriptionResponse_EncodeableType,*/
    OpcUa_Server_BeginCreateSubscription,
    (OpcUa_PfnInvokeService*)OpcUa_ServerApi_CreateSubscription
};
```
but i do'nt know how to send data for the Subscription

Lama1349, 2017-05-10:
To get the subscription/monitored items running you must support at minimum the "Embedded DataChange Subscription Server Facet" (check for this [(https://opcfoundation-onlineapplications.org/profilereporting/)]).
Which services and it's behaviour the server must support you can see in the UA Spec. Part 4, chapter 5.12 and 5.13. I guess in sum the sample server here must be extend with approximately 10 services and the associated behaviour.

loujunang, 2017-09-05:
I have the same question, can you provide a detailed case?

Lama1349, 2017-09-05:
What did you mean which a detailes case?
For example which UA Services exactly?

loujunang, 2017-09-05:
Examples of subscription functions

loujunang, 2017-09-05:
`// OpcUa_ServiceType  dummy_CreatSubscription =
// 	{	OpcUaId_CreateSubscriptionRequest,
// 		OpcUa_Null, /*&OpcUa_CreateSubscriptionResponse_EncodeableType,*/
// 		(OpcUa_PfnBeginInvokeService*)OpcUa_Server_BeginCreateSubscription,
// 		(OpcUa_PfnInvokeService*)my_CreateSubscription};

struct _OpcUa_ServiceType my_CreatSubscription_ServiceType =
{
	OpcUaId_CreateSubscriptionRequest,
	OpcUa_Null,
	OpcUa_Server_BeginCreateSubscription,
	(OpcUa_PfnInvokeService*)my_CreateSubscription
};

struct _OpcUa_ServiceType my_CreateMonitoredItems_ServiceType =
{
	OpcUaId_CreateMonitoredItemsRequest,
	OpcUa_Null,
	OpcUa_Server_BeginCreateMonitoredItems,
	(OpcUa_PfnInvokeService*)my_CreateMonitoredItems
};

struct _OpcUa_ServiceType my_DeleteSubscriptions_ServiceType =
{
	OpcUaId_DeleteSubscriptionsRequest,
	OpcUa_Null,
	OpcUa_Server_BeginDeleteSubscriptions,
	(OpcUa_PfnInvokeService*)my_DeleteSubscriptions
};

/** @brief All supported services. */
OpcUa_ServiceType*  UaTestServer_SupportedServices[] =
{
    &OTServer_ServiceGetEndpoints,
    &ServiceCreatSession,
    &ActivateSession,
    &CloseSession,
    &my_Browse_ServiceType,
    &my_Read_ServiceType,
    &my_BrowseNext_ServiceType,
    &my_FindServers_ServiceType,
    &my_CreatSubscription_ServiceType,
	&my_CreateMonitoredItems_ServiceType,
	&my_DeleteSubscriptions_ServiceType,
    OpcUa_Null
};`

CreateSubsription succeeded
CreateMonitoredItems failed 
Deleting subscription failed

Lama1349, 2017-09-06:
I cannot give some example since we only use the OPC Foundation stack with an own server implementation which is not compatible to the given sample server. But if you look into the UA Spec. Part 4, V1.03, chapter 5.13.1.2 describes the Subscription state table, chapter 5.13.1.3 describes the required variables and chapter 5.13.1.4 described the required functions for the Subscription/MonitoredItem handling.

alexsmn, 2017-09-06:
You may see a brief example here https://github.com/alexsmn/opcuapp/blob/master/opcuapp/server/subscription.cpp
(Sorry for the shameless promotion though.)

Anyway, any implementation of subscription state transition is quite welcome, since even its' description in Part 4 - Services is quite complicated. I would also appreciate some code examples regarding monitored item sampling, queuing and triggering.