OPCFoundation/UA-AnsiC-Legacy #46: Secure connection can not communicate

iPuerto, 2017-05-11:
Hello,

I tried secure connetion using AnsiC stack (UA Server). But, the secure connection can not communicate because error occurs at 509 line of 'opcua_p_openssl_rsa.c'. Does this stack support secure connection?

Test tools(Client): UaExpert, OPC-F Sample Client
Security mode: SignAndEncrypt
Security policy: Basic128Rsa15
Authentication Settings: Anonymous

Development tools: Visual Studio 2017
Execution environment: Windows10

bernd-edlinger, 2017-07-19:
Hi,

the stack itself does support security in principle.
But the sample does only implement a non-secure endpoint.

iPuerto, 2017-07-19:
Hi,

Thank you for your reply. I understood. I will try.

closed 2017-07-19 by iPuerto