OPCFoundation/UA-AnsiC-Legacy #52: Activating a secure connection

sheldonreddy, 2017-08-02:
Greetings, 

I am currently trying to create a secure ua client to do a basic connect, read and disconnect from the OPC Foundation Sample UA Server. 

I am using the AnsiC Stack as the basis for my work. Within the **_opcua_clientapi.c file_** - there is a function called _**OpcUa_ClientApi_ActivateSession()**_ 

This function requires the clientSignature as one of the parameters required to activate the session. The .NET sample code has the following within the **session.cs** file:

```
byte[] dataToSign = Utils.Append(endpoint.ServerCertificate, m_serverNonce);
SignatureData clientSignature = SecurityPolicies.Sign(m_instanceCertificate, endpoint.SecurityPolicyUri, dataToSign);
```
I am trying to achieve the equivalent of the second function above using the AnsiC stack however I can't seem to find a function within the stack to assist me. 

I would really appreciate any suggestions or advice. I have used the sample Publisher code however this is not handled there for some reason?

TIA
Sheldon

bernd-edlinger, 2017-08-04:
You should be able to create the necessary signature with `OpcUa_Crypto_AsymmetricSign`.

sheldonreddy, 2017-08-04:
Hi Bernd, 

I have found this function and am busy attempting to pass the correct parameters. 

Thank you for your assistance.
I will close if I have no further questions. 

Thank you

sheldonreddy, 2017-08-04:
Hi Bernd, 

Thank you, I did find this function a day ago, I am just busy trying to fix and error with the first parameter expected by the function.

The function `OpcUa_Crypto_AsymmetricSign` operates as follows:

```
OPCUA_EXPORT OpcUa_StatusCode OpcUa_Crypto_AsymmetricSign(
    struct _OpcUa_CryptoProvider*    a_pProvider,
    OpcUa_ByteString                 a_data,
    OpcUa_Key*                       a_privateKey,    
    OpcUa_ByteString*                a_pSignature)
{
   //The function code exists here

}
```
I am having trouble figuring out what the first parameter should be in my function call to `OpcUa_Crypto_AsymmetricSign()`

Would you be able to assist me in determining what the first parameter should be?

Thanks Bernd

bernd-edlinger, 2017-08-07:
`OpcUa_CryptoProvider_Create` does that.

sheldonreddy, 2017-08-07:
Fantastic, I just want to confirm, 

To generate the `a_privateKey` parameter of type `OpcUa_key `- one would use the `OpcUa_Crypto_GetPrivateKeyFromCert` function?

sheldonreddy, 2017-08-07:
Thanks Bernd. 

I am just struggling to figure out how to get the privatekey into the OpcUa_Key format as I use it as a bytestring every where else.

bernd-edlinger, 2017-08-07:
The OpcUa_Key contains a `Key` which is the private Key in DER format,
and a `Type` which has to be set to the value
`OpcUa_Crypto_KeyType_Rsa_Private`

sheldonreddy, 2017-08-07:
I actually came across that a little while ago after some digging around. Good to know I am on the right track.

 Thank you for your assistance. I am setting it up right now 👍

sheldonreddy, 2017-08-08:
@bernd-edlinger I have used the following function to load the private key from the PKI. 

`OpcUa_P_OpenSSL_RSA_LoadPrivateKeyFromFile()`

I understand one of the parameters to this function is the location of the privatekey within the pki. Is the following a valid setup for the variable?

`/private/privatekeyname.pem`

The PKI folder has the Private folder within it where the private key resides.

I have set the` requested encoding format` parameter as PEM for the function.

sheldonreddy, 2017-08-08:
The issue I am facing is loading the `privateKey` into a variable of type` .DER `or `.PEM` so that I can use it to create the variable of type `OpcUa_Key` to be used in the `OpcUa_CryptoProvider_Create()` function

The function `OpcUa_P_OpenSSL_RSA_LoadPrivateKeyFromFile()` returns the `privatekey` as a `bytestring`.

bernd-edlinger, 2017-08-08:
You should not directly call this function but instead use the
OpcUa_PKIProvider.LoadPrivateKeyFromFile function pointer for that.

Actually I am a bit unhappy with the way the LoadPrivateKey function works.
And I am on the way to change the interface to use a OpcUa_Key,
and to support other key types than RSA.

sheldonreddy, 2017-08-08:
That's actually great news @bernd-edlinger - I do feel it would allow a lot more flexibility for developers.

Can I assume the  `OpcUa_Key_ClearHandle   fpClearHandle` can be null for the `OpcUa_Key pvtKey` variable?

sheldonreddy, 2017-08-08:
That's great new @bernd-edlinger - I feel it will allow for a lot more flexibility for OPC Developers. Thanks for looking into this. 

I have attempted to call this function from my main function however I am experiencing an error suggest `a_pPKI `is not initialized.

```
        OpcUa_PKIProvider* a_pPKI;
      
	a_pPKI->LoadPrivateKeyFromFile(a_privateKeyFile, OpcUa_Crypto_Encoding_PEM, password, a_pPrivateKey);

```
Is this function call correct?

bernd-edlinger, 2017-08-08:
you need to call `OpcUa_PKIProvider_Create` first.

bernd-edlinger, 2017-08-08:
Use always be the full path.  But it should not crash if the file could not be opened.

sheldonreddy, 2017-08-10:
I have managed to fix the issue. 

with regards to the `ClientSignature` variable required by the `ActivatSession` in the `ClientAPI`

I attempt to create it as follows:

```
uStatus = OpcUa_String_AttachReadOnly(securityURIstring, OpcUa_SecurityPolicy_Basic128Rsa15);					

OpcUa_CryptoProvider AsymmetricSecurityURI;
uStatus = OpcUa_CryptoProvider_Create(securityURIstring, &AsymmetricSecurityURI);				

uStatus = OpcUa_Crypto_AsymmetricSign(&AsymmetricSecurityURI, a_pSession->ClientSig, &pvtKey, &myClientSignature.Signature);

myClientSignature.Algorithm = *securityURIstring;
```
The issue is with the `AsymmetricSecurityURI`. The questions:
When using the  `OpcUa_CryptoProvider_Create()` function - am I passing `AsymmetricSecurityURI` correctly ? in terms of type. Prior to the  `OpcUa_Crypto_AsymmetricSign() `executing - the `AsymmetricSecurityURI` is empty and hence the function does not execute correctly. 

The issue is definitely with the ` AsymmetricSecurityURI` variable. Looking within the stack, and in particular, the implementation of `OpcUa_CryptoProvider_Create()` - it requires a `OpcUa_CryptoProvider `variable to create the `CryptoProvider` to be used in the `OpcUa_Crypto_AsymmetricSign()` function. Therefore I created the variable ` AsymmetricSecurityURI` and passed it to the function.

What would you suggest I do to ensure the functions above execute correctly?
Thanks @bernd-edlinger

sheldonreddy, 2017-08-16:
Hi Bernd, 

The `OpcUa_ClientApi_ActivateSession() `requires a `clientSoftwareCertificate` parameter of type `OpcUa_SignedSoftwareCertificate*`

The structure of this type has two bytestrings - one for `CertificateData` and one for `Signature.`

The Signature variable is defined in PART4 of the OPC UA Specification as `the signature created by the issuer`. 

I used the OPC Foundation UA Configuration Tool to create the `clientCertificate` and `privateKey`. I assume the UA Configuration Tool is then the issuer? 

My question is what do I populate the Signature field within the `clientSoftwareCertificate` structure?

cacamille3, 2017-09-15:
I think you should ignore the ClientSignature in an ActivateSessionRequest.
Currently the specification described it as "Reserved for future use."

closed 2018-08-02 by bernd-edlinger