OPCFoundation/UA-AnsiC-Legacy #54: Ansi C memory leak with OpcUa_ClientApi_Read

andreabatcost, 2017-09-14:
I have a client (based on the example https://github.com/OPCFoundation/UA-AnsiC/tree/prototyping/AnsiCSamplePublisher)
that loops forever with a call to OpcUa_ClientApi_Read to read only one item serverTimeNodeID = 2258  and I observe the memory rising continuosly.
Is there a memory leak in the stack with OpcUa_ClientApi_Read ?

Thank you
Andrea

cacamille3, 2017-09-14:
I think both pointer `pResults `and `pDiagnosticInfos` passed in the `OpcUa_ClientApi_Read` are leaking.

- `pDiagnosticInfos:  `should be cleared and freed in `Publisher_ReadNode` 

- `pResults: `is assigned to `*pValueRead` the argument of `Publisher_ReadNode`, so this one shall be cleared and freed in `Main_Publisher`

andreabatcost, 2017-09-14:
Thank you very much.
It was just an OpcUa_Free(value); missing in my Main_Publisher.
Now the memory is stable.
Best regards,
Andrea

closed 2017-09-14 by andreabatcost