OPCFoundation/UA-AnsiC-Legacy #61: Memory/resource leak in opcua_tcplistener.c

eckes, 2017-09-25:
https://github.com/OPCFoundation/UA-AnsiC/blob/0ebcdfbb4d520066cf6559542ce42b24878d3f96/Stack/transport/tcp/opcua_tcplistener.c#L1810

In `OpcUa_TcpListener_ConnectionDeleteCallback()`, there's a call to `OPCUA_P_SOCKET_CLOSE()` for closing the porting layer socket when a connection gets deleted.

Unfortunately, this call is commented out for whatever reason so any resources held by `a_pTcpConnection->Socket` including the socket's memory itself are leaking.

This was already reported in Mantis as issue [2674](https://opcfoundation-onlineapplications.org/mantis/view.php?id=2674) but received no attention.

bernd-edlinger, 2017-09-25:
When this callback is invoked the socket manager is supposed to be deleted already.
The socket manager is supposed to close all open sockets.
So I don't see what's the problem here.

eckes, 2017-09-25:
Bernd, thanks for the quick reply.

Maybe things *should* work as you described them but when `OPCUA_MULTITHREADED` is set to `OPCUA_CONFIG_NO`, the `OPCUA_P_SOCKETMANAGER_DELETE` function isn't called at all:
https://github.com/OPCFoundation/UA-AnsiC/blob/0ebcdfbb4d520066cf6559542ce42b24878d3f96/Stack/transport/tcp/opcua_tcplistener.c#L1845-L1854

Your described behavior implies that the socket manager tracks its issued sockets which introduces additional management effort in the socket manager.

If the `OPCUA_P_SOCKETMANAGER_DELETE` function would be called, the socket manager could decide in the implementation what to do.

Wouldn't it be a good idea to call `OpcUa_TcpListener_ConnectionManager_RemoveConnections` **before** the socket manager gets deleted? The signed in callback (`OpcUa_TcpListener_ConnectionDeleteCallback`) is responsible for cleaning up a socket held in a given `OpcUa_TcpListener_Connection`. Things can get odd when the socket manager is cleaned up before the sockets were cleaned up IMHO.

bernd-edlinger, 2017-09-25:
Please see this statement in the ChangeLog.txt:

```
                - 3410 OPCUA_MULTITHREADED: CONFIG_NO is supported,
                                 but you must call OpcUa_SocketManager_Create(OpcUa_Null,0,0)
                                 before OpcUa_Endpoint_Open, and you must call
                                 OpcUa_SocketManager_Delete(OpcUa_Null) between
                                 OpcUa_Endpoint_Close and Opcua_Endpoint_Delete.
```

bernd-edlinger, 2017-09-25:
And moreover, closing the sockets on delete is the expected behaviour of the socket manager.
If you use a different socket manager with different behaviour I cannot help...

closed 2018-08-02 by bernd-edlinger