OPCFoundation/UA-AnsiC-Legacy #62: Option to choose IPv4/IPv6 address during  OpcUa_Channel_Connect

srinarasi, 2017-09-26:
When we call OpcUa_Channel_Connect, I see that we call getaddrinfo() and pick the first address in the list for creating the client socket.

In my case, the first address in the list is always IPv6 and the server I'm testing with doesn't seem to work well with IPv6.

Here are my questions:

1) Is there any way to choose the IP address family while trying to connect to the server? 

2) If not, I'd like to extend the OpcUa_Channel_Connect interface with an option to choose IPv4/IPv6 addresses. Would it be a good idea? If it is, I can implement it send you a pull request.

bernd-edlinger, 2017-09-26:
Actually I am not overly happy with the name resolution in a low-level stack function,
because this is the *only* blocking function in the whole stack, everything else is done
in a non-blocking way.  The name resolution could easily block for around a minute or so.

Therefore I would do the name resolution in the application, and try to connect to one
numeric IPv4 address and one numeric IPv6 address in *parallel*.  When both connects
succeed, use the faster one and drop the other one.

When numeric IP-addresses are used, it is clear for the stack, which address family to use,
and there are also no delays in the name resolution.

srinarasi, 2017-10-09:
Thanks for the response. I implemented name resolution in the application.

closed 2018-08-02 by bernd-edlinger