OPCFoundation/UA-AnsiC-Legacy #64: Stack does not compile as C++ code

MarkusErlmann, 2017-10-16:
Currently, the stack cannot be compiled as C++ code because of missing casts after allocations (trigger C2440 with Visual Studio) and comparisons against `== OpcUa_True` (trigger C4805 with Visual Studio).

These missing casts and wrong comparisons are fixed in PR #63

bernd-edlinger, 2017-10-19:
But this is the ANSI C Stack.

MarkusErlmann, 2017-10-20:
Yes, of course it is.

But with a *handfull* of trivial changes it can also be compiled as C++ code with stricter type checking which I would consider being a good thing.

For example, it helps you spot errors like this one:
https://github.com/OPCFoundation/UA-AnsiC/pull/63/files#diff-c3842f8a57f7568c0862693fd9902499L671

bernd-edlinger, 2017-10-20:
Hmm, that is more like a bug-fix and has nothing to do with trivial changes.

bernd-edlinger, 2017-10-20:
BTW: The warnings on the comparisons against `== OpcUa_True` indicate a problem with
the platform layer you use.
`OpcUa_True` and `OpcUa_False` are by definition the correct values for a `OpcUa_Boolean`.

closed 2018-08-02 by bernd-edlinger