OPCFoundation/UA-AnsiC-Legacy #66: Potential memory leak in OpcUa_P_OpenSSL_X509_SelfSigned_Custom_Create

MarkusErlmann, 2017-10-16:
When `OPENSSL_realloc` returns a null pointer, the original memory of `pSerialNumber->Data` is lost.

A fix is available in #65.

bernd-edlinger, 2017-10-19:
Thanks, but this is a way too complicated fix, because `pSerialNumber->Data` is always `NULL`.
Maybe just using `OPENSSL_malloc` would be more appropriate.

However I see a different issue here:

That is IF the GUID starts with 0x00, 0x00..0x7F the generated certificate uses an invalid ASN1
encoded serialNumber. One out of 512 created certificates have this anomaly.

I tried to create certificates that use such a serialNumber, but they seem to work with OpenSSL,
but it is possible that there are IOP issues with other libraries.

MarkusErlmann, 2017-10-19:
@bernd-edlinger yes, if `pSerialNumber->Data` is always `NULL` here, I do also think that `OPENSSL_malloc` would be the better choice here.

Regarding the GUID issue: what would be the solution here?

bernd-edlinger, 2017-10-19:
Are you affected by this issue?
I think it would be sufficient to look at the first byte and change it to 1 when it is 0.
I will put that on my TODO list.

MarkusErlmann, 2017-10-19:
@bernd-edlinger No, I'm currently not affected. Fix sounds reasonable.

bernd-edlinger, 2017-10-20:
fixed on trunk.

closed 2017-10-20 by bernd-edlinger