OPCFoundation/UA-AnsiC-Legacy #72: CMake build system should support the "install" target and, ideally, various packaging formats

DavidZemon, 2017-10-30:
The CMake build scripts currently in place do not provide an `install` target. For a C library, that seems like an obvious enough problem.

Once the install target is created, CMake should be used to its fullest potential and support for CPack should be added. This allows for easy creation of `.tar.gz`, `.deb`, `.rpm`, `.exe` and other installation formats.

DavidZemon, 2017-10-30:
If this is something that the owners of the project are interested in adding, I'm more than happy to submit a pull request for this feature.

ksk-, 2017-11-07:
Good point! It is very necessary.

LeSpocky, 2018-07-24:
I have a [personal branch](https://github.com/LeSpocky/UA-AnsiC/tree/build) for that, but I think it's not ready for inclusion, because I can not test on WIN32. Is there interest in making a pull request from that?

LeSpocky, 2018-08-02:
Leaving the ABI compat discussion aside, this:

> install the application which is linked to the stack that was used in building the application would be my preference.

In reverse conclusion that would mean you must import the stack code in your application project, e.g. with git submodules or something similar, to build and link it. Irrelevant whether you do static or dynamic linking, building this project on its own and linking from your separate application would not be possible, because that would require _installing_ the (static/dynamic) library. To link the stack you somehow need the stack's CMake target, either by inclusion as "subproject" or by `find_package()` and importing the _installed_ target export files. (Yes, just linking the installed files would be possible, but that's not the modern CMake way. You might have a look at my patches how it could look in the end.)

bernd-edlinger, 2018-08-02:
I don't really know enough CMake to judge.
In my projects I use hierarchical make files, Stack is just one of them, but the compilation
results of the Stack are not necessarily deployed.

DavidZemon, 2018-08-06:
Providing an `install` target for a library has nothing to do with installing the library on an end-user's computer. It's about easy development of applications that use the library. For instance, if you want to use zlib, I don't go download the zlib source code and compile it myself. I type `sudo apt-get install zlib1g-dev` and I have "installed" my library along with header files. Now, it may not be worth setting up a PPA repository for debian-based distros, but there's nothing saying we can't provide pre-compiled binaries in various packaging formats, such as `.tar.gz`, `.tar.bz2`, `.deb`, `.rpm`, and `.exe`. That makes it much easier for anyone that wants to link against the UA-AnsiC library.

And even if we don't provide the exact binary format (maybe we provide a `.tar.gz` for Linux x86-64, but some company/group wants to use this on a Raspberry Pi), at least our build system will support the packaging format. In that case, one person in that company or group will build from source for Raspberry Pi, and then provide the binary package to all other members of that company/group.

Every C/C++ project - no matter if it is an application or a library - should provide an `install` target. And because this project uses CMake, we can take advantage of CPack to provide packages as well.