OPCFoundation/UA-AnsiC-Legacy #80: Possible null pointer dereferencing in OpcUa_String_Initialize()

cacamille3, 2017-12-07:
Currently the argument `OpcUa_String* a_pString` is not checked with
`OpcUa_ReturnErrorIfArgumentNull(a_pString); `

This could result to a segmentation fault in OPC UA Client or Server using this function which is available through _./core/opcua_string.h_

```
OpcUa_StatusCode OpcUa_String_Initialize( /*  bi */ OpcUa_String* a_pString )
{
    OpcUa_pStringInternal   pStringInt  = (OpcUa_pStringInternal)   a_pString;

    pStringInt->uMagic          = OpcUa_uiMagic;
    pStringInt->bFreeSecondMem  = OpcUa_True;
    /*pStringInt->uReserved       = 0;*/
    pStringInt->uLength         = 0;
    pStringInt->strContent      = OpcUa_Null;

    return OpcUa_Good;
}
```

cacamille3, 2017-12-08:
Pull Request #81

bernd-edlinger, 2017-12-08:
Actually I am not sure if OpcUa_String_Initialize should better return Void,
like all other OpcUa_&lt;DataType&gt;_Initialize functions.

Right now the "error tolerance" of the OpcUa_String objects makes more
problems than it helps.

cacamille3, 2017-12-08:
What do you mean with "error tolerance" ? 
But yes it might be a better idea to follow the pattern of other Builtin types

bernd-edlinger, 2017-12-08:
> What do you mean with "error tolerance" ?

I mean if an OpcUa_String_Initialize has been forgotten, the following
OpcUa_String_AttachXxx or OpcUa_String_Clear should crash immediately,
but they rarely do.

I for one would prefer if they aborted every time rather than silently ignoring
the problem until a certain bit pattern is on the uninitialized object.

cacamille3, 2017-12-08:
Yes totally agree, I have notified this scenario in my OPC UA Server as well...

bernd-edlinger, 2018-06-25:
Fixed by f4d0d18.

closed 2018-06-25 by bernd-edlinger