OPCFoundation/UA-AnsiC-Legacy #104: Windows10 + CMake + mingw32-make Fail!

yanniupai, 2018-05-31:
The CMake is ok, like below:
```
The C compiler identification is GNU 5.3.0
The CXX compiler identification is GNU 5.3.0
Check for working C compiler: C:/Qt/Qt5.11.0/Tools/mingw530_32/bin/gcc.exe
Check for working C compiler: C:/Qt/Qt5.11.0/Tools/mingw530_32/bin/gcc.exe -- works
Detecting C compiler ABI info
Detecting C compiler ABI info - done
Detecting C compile features
Detecting C compile features - done
Check for working CXX compiler: C:/Qt/Qt5.11.0/Tools/mingw530_32/bin/g++.exe
Check for working CXX compiler: C:/Qt/Qt5.11.0/Tools/mingw530_32/bin/g++.exe -- works
Detecting CXX compiler ABI info
Detecting CXX compiler ABI info - done
Detecting CXX compile features
Detecting CXX compile features - done
Found OpenSSL: D:/Professional/OPC/UA-AnsiC-master/openssl/lib/libeay32.lib (found version "1.0.2j") 
Configuring done
Generating done
```
Next, I buile it with MinGW32-Make, the resualt like below:
```
D:\Professional\OPC\UA-AnsiC-master\Build>mingw32-make
Scanning dependencies of target uastack
[  1%] Building C object Stack/CMakeFiles/uastack.dir/core/opcua_buffer.c.obj
[  2%] Building C object Stack/CMakeFiles/uastack.dir/core/opcua_core.c.obj
[  3%] Building C object Stack/CMakeFiles/uastack.dir/core/opcua_datetime.c.obj
[  4%] Building C object Stack/CMakeFiles/uastack.dir/core/opcua_guid.c.obj
[  5%] Building C object Stack/CMakeFiles/uastack.dir/core/opcua_list.c.obj
[  6%] Building C object Stack/CMakeFiles/uastack.dir/core/opcua_memory.c.obj
[  7%] Building C object Stack/CMakeFiles/uastack.dir/core/opcua_memorystream.c.obj
[  8%] Building C object Stack/CMakeFiles/uastack.dir/core/opcua_proxystub.c.obj
[ 10%] Building C object Stack/CMakeFiles/uastack.dir/core/opcua_string.c.obj
[ 11%] Building C object Stack/CMakeFiles/uastack.dir/core/opcua_thread.c.obj
[ 12%] Building C object Stack/CMakeFiles/uastack.dir/core/opcua_threadpool.c.obj
[ 13%] Building C object Stack/CMakeFiles/uastack.dir/core/opcua_timer.c.obj
[ 14%] Building C object Stack/CMakeFiles/uastack.dir/core/opcua_trace.c.obj
[ 15%] Building C object Stack/CMakeFiles/uastack.dir/core/opcua_utilities.c.obj
[ 16%] Building C object Stack/CMakeFiles/uastack.dir/proxystub/clientproxy/opcua_asynccallstate.c.obj
[ 17%] Building C object Stack/CMakeFiles/uastack.dir/proxystub/clientproxy/opcua_channel.c.obj
[ 18%] Building C object Stack/CMakeFiles/uastack.dir/proxystub/clientproxy/opcua_clientapi.c.obj
[ 20%] Building C object Stack/CMakeFiles/uastack.dir/proxystub/serverstub/opcua_endpoint.c.obj
[ 21%] Building C object Stack/CMakeFiles/uastack.dir/proxystub/serverstub/opcua_serverapi.c.obj
[ 22%] Building C object Stack/CMakeFiles/uastack.dir/proxystub/serverstub/opcua_servicetable.c.obj
[ 23%] Building C object Stack/CMakeFiles/uastack.dir/securechannel/opcua_secureconnection.c.obj
[ 24%] Building C object Stack/CMakeFiles/uastack.dir/securechannel/opcua_securelistener.c.obj
[ 25%] Building C object Stack/CMakeFiles/uastack.dir/securechannel/opcua_securelistener_channelmanager.c.obj
[ 26%] Building C object Stack/CMakeFiles/uastack.dir/securechannel/opcua_securelistener_policymanager.c.obj
[ 27%] Building C object Stack/CMakeFiles/uastack.dir/securechannel/opcua_securestream.c.obj
[ 28%] Building C object Stack/CMakeFiles/uastack.dir/securechannel/opcua_soapsecurechannel.c.obj
[ 30%] Building C object Stack/CMakeFiles/uastack.dir/securechannel/opcua_tcpsecurechannel.c.obj
[ 31%] Building C object Stack/CMakeFiles/uastack.dir/stackcore/opcua_binarydecoder.c.obj
[ 32%] Building C object Stack/CMakeFiles/uastack.dir/stackcore/opcua_binaryencoder.c.obj
[ 33%] Building C object Stack/CMakeFiles/uastack.dir/stackcore/opcua_builtintypes.c.obj
[ 34%] Building C object Stack/CMakeFiles/uastack.dir/stackcore/opcua_connection.c.obj
[ 35%] Building C object Stack/CMakeFiles/uastack.dir/stackcore/opcua_crypto.c.obj
[ 36%] Building C object Stack/CMakeFiles/uastack.dir/stackcore/opcua_decoder.c.obj
[ 37%] Building C object Stack/CMakeFiles/uastack.dir/stackcore/opcua_encodeableobject.c.obj
[ 38%] Building C object Stack/CMakeFiles/uastack.dir/stackcore/opcua_encoder.c.obj
[ 40%] Building C object Stack/CMakeFiles/uastack.dir/stackcore/opcua_enumeratedtype.c.obj
[ 41%] Building C object Stack/CMakeFiles/uastack.dir/stackcore/opcua_extensionobject.c.obj
[ 42%] Building C object Stack/CMakeFiles/uastack.dir/stackcore/opcua_listener.c.obj
[ 43%] Building C object Stack/CMakeFiles/uastack.dir/stackcore/opcua_messagecontext.c.obj
[ 44%] Building C object Stack/CMakeFiles/uastack.dir/stackcore/opcua_pki.c.obj
[ 45%] Building C object Stack/CMakeFiles/uastack.dir/stackcore/opcua_securechannel.c.obj
[ 46%] Building C object Stack/CMakeFiles/uastack.dir/stackcore/opcua_stream.c.obj
[ 47%] Building C object Stack/CMakeFiles/uastack.dir/stackcore/opcua_stringtable.c.obj
[ 48%] Building C object Stack/CMakeFiles/uastack.dir/stackcore/opcua_types.c.obj
[ 50%] Building C object Stack/CMakeFiles/uastack.dir/transport/https/opcua_httpsconnection.c.obj
[ 51%] Building C object Stack/CMakeFiles/uastack.dir/transport/https/opcua_httpslistener.c.obj
[ 52%] Building C object Stack/CMakeFiles/uastack.dir/transport/https/opcua_httpslistener_connectionmanager.c.obj
[ 53%] Building C object Stack/CMakeFiles/uastack.dir/transport/https/opcua_httpsstream.c.obj
[ 54%] Building C object Stack/CMakeFiles/uastack.dir/transport/https/opcua_https_internal.c.obj
[ 55%] Building C object Stack/CMakeFiles/uastack.dir/transport/tcp/opcua_tcpconnection.c.obj
[ 56%] Building C object Stack/CMakeFiles/uastack.dir/transport/tcp/opcua_tcplistener.c.obj
[ 57%] Building C object Stack/CMakeFiles/uastack.dir/transport/tcp/opcua_tcplistener_connectionmanager.c.obj
[ 58%] Building C object Stack/CMakeFiles/uastack.dir/transport/tcp/opcua_tcpstream.c.obj
[ 60%] Building C object Stack/CMakeFiles/uastack.dir/platforms/win32/opcua_p_binary.c.obj
[ 61%] Building C object Stack/CMakeFiles/uastack.dir/platforms/win32/opcua_p_cryptofactory.c.obj
[ 62%] Building C object Stack/CMakeFiles/uastack.dir/platforms/win32/opcua_p_datetime.c.obj
[ 63%] Building C object Stack/CMakeFiles/uastack.dir/platforms/win32/opcua_p_guid.c.obj
[ 64%] Building C object Stack/CMakeFiles/uastack.dir/platforms/win32/opcua_p_interface.c.obj
[ 65%] Building C object Stack/CMakeFiles/uastack.dir/platforms/win32/opcua_p_internal.c.obj
D:\Professional\OPC\UA-AnsiC-master\Stack\platforms\win32\opcua_p_internal.c: In function 'OpcUa_P_GetDateTimeDiff':
D:\Professional\OPC\UA-AnsiC-master\Stack\platforms\win32\opcua_p_internal.c:71:61: error: invalid suffix "i64" on integer constant
     a_pResult->dwHighDateTime = (OpcUa_UInt32)((ullResult & 0xFFFFFFFF00000000i64) >> 32);
                                                             ^
Stack\CMakeFiles\uastack.dir\build.make:874: recipe for target 'Stack/CMakeFiles/uastack.dir/platforms/win32/opcua_p_internal.c.obj' failed
mingw32-make[2]: *** [Stack/CMakeFiles/uastack.dir/platforms/win32/opcua_p_internal.c.obj] Error 1
CMakeFiles\Makefile2:84: recipe for target 'Stack/CMakeFiles/uastack.dir/all' failed
mingw32-make[1]: *** [Stack/CMakeFiles/uastack.dir/all] Error 2
Makefile:93: recipe for target 'all' failed
mingw32-make: *** [all] Error 2

```

Do your give me some idear? Thanks

thradams, 2018-05-31:
i64 is non standard suffix.
Maybe ll instead.

yanniupai, 2018-05-31:
thradams, Is it linke this
`0xFFFFFFFF00000000l64`

closed 2018-05-31 by yanniupai