OPCFoundation/UA-AnsiC-Legacy #115: Windows10 + Cmake + ming32-make build UA-AnsiC fail

yanniupai, 2018-06-26:
```
../lib/libuastack.a(opcua_p_openssl.c.obj):opcua_p_openssl.c:(.text+0xc5): undefined reference to `CRYPTO_set_locking_callback'
../lib/libuastack.a(opcua_p_openssl.c.obj):opcua_p_openssl.c:(.text+0xca): undefined reference to `OPENSSL_add_all_algorithms_noconf'
../lib/libuastack.a(opcua_p_openssl.c.obj):opcua_p_openssl.c:(.text+0xcf): undefined reference to `RAND_screen'
../lib/libuastack.a(opcua_p_openssl.c.obj):opcua_p_openssl.c:(.text+0xd4): undefined reference to `SSL_library_init'
../lib/libuastack.a(opcua_p_openssl.c.obj):opcua_p_openssl.c:(.text+0xd9): undefined reference to `SSL_load_error_strings'
../lib/libuastack.a(opcua_p_openssl.c.obj):opcua_p_openssl.c:(.text+0xeb): undefined reference to `SSL_COMP_free_compression_methods'
../lib/libuastack.a(opcua_p_openssl.c.obj):opcua_p_openssl.c:(.text+0xf0): undefined reference to `EVP_cleanup'
../lib/libuastack.a(opcua_p_openssl.c.obj):opcua_p_openssl.c:(.text+0xf5): undefined reference to `CRYPTO_cleanup_all_ex_data'
../lib/libuastack.a(opcua_p_openssl.c.obj):opcua_p_openssl.c:(.text+0x101): undefined reference to `ERR_remove_state'
../lib/libuastack.a(opcua_p_openssl.c.obj):opcua_p_openssl.c:(.text+0x106): undefined reference to `ERR_free_strings'
../lib/libuastack.a(opcua_p_openssl.c.obj):opcua_p_openssl.c:(.text+0x112): undefined reference to `CRYPTO_set_locking_callback'
```

The openssl was not linking to UA-AnsiC, how to link it?

bernd-edlinger, 2018-06-27:
mingw32 is not a supported platform

yanniupai, 2018-06-28:
I used CMake to cross-compiling UA-AnsiC，it is ok；But next Mingw32-make it fail，and the openssl is not linked，if it linking，that is ok.

bernd-edlinger, 2018-06-28:
Yes, maybe you look at the build_win32.bat that might be able to build openssl.

yanniupai, 2018-07-08:
I build openssl ok with MSYS2/mingw, but i don't known how to link UA-AnsiC and openssl, and how to build the UA-AnsiC, Anyone has a idear?
       The Visual C++ is not a multi-platform IDE, But OPC-UA is a multi-platform lib. so we want build it with mingw or QT. Do you help me?

bernd-edlinger, 2018-07-09:
Please look at the README.md:
```
## Building the Stack

### Windows

Open the Visual Studio Command Shell .
Make sure that perl installed on the build machine and the path executable is available in the path.  Any perl will do, even cygwin.
Download and extract the latest openssl-1.0.2 source tar ball to the stack root folder.
Then cd to the root folder and execute build_win32.bat or build_win64.bat depending on your target architecture.
This will automatically build openssl, the OPC UA Stack and the sample server.
Dependencies are not supported, it will always be a full build.

```

bernd-edlinger, 2018-07-23:
Yes, but only if you install a pre-compiled openssl.
Sorry I cannot help you.

closed 2018-07-23 by bernd-edlinger