OPCFoundation/UA-AnsiC-Legacy #120: What is the way to connect from OPC UA Client with UserName and Password?

sanny32, 2018-07-27:
I'm try to connect to the external Opc Ua Server with UA-AnsiC SDK.
I'm create `OpcUa_UserNameIdentityToken` and pass the username to the `UserName` field.
The problem to pass a user password to the `Password` field. A code comment says 

> This parameter shall be encrypted with the Server's Certificate using the algorithm specified by the
 SecurityPolicy

What is the way to encrypt password with servrer's certificate using algorithm by the security policy? What functions i should to use?

cacamille3, 2018-07-27:
You shall use the information returned by the Server in its CreateSession response
 - The Certificate of the Server
 - The Server Nonce 
 - The SecurityPolicy to use for the UserIdentityToken you want to use (shall be the one with the PolicyId 'UserName')

Refer to[ OPC UA Specification Part 4](https://opcfoundation.org/developer-tools/specifications-unified-architecture/part-4-services/) - 7.36 UserIdentityToken parameters to know exactly how to encode the UserNameIdentityToken

sanny32, 2018-07-28:
@cacamille3 thanks for help. 
After some tests i have write this code to encrypt the user password.

    OpcUa_StatusCode EncrypPassword(const chr* utf8password, const char* securityProfileUri, OpcUa_ByteString* serverCertificate, OpcUa_ByteString serverNonce, OpcUa_ByteString* encryptedData)
    {
        OpcUa_ReturnErrorIfArgumentNull(utf8password);
        OpcUa_ReturnErrorIfArgumentNull(securityProfileUri);
        OpcUa_ReturnErrorIfArgumentNull(serverCertificate);
        OpcUa_ReturnErrorIfArgumentNull(serverNonce);
        OpcUa_ReturnErrorIfArgumentNull(encryptedData);

        OpcUa_Int32 dataLength;
        OpcUa_Key publicKey;
        OpcUa_CryptoProvider cp;
        OpcUa_Int32 utf8PasswordLen = OpcUa_StrLen(utf8password);
        OpcUa_StatusCode uStatus = OpcUa_Good;

        OpcUa_ByteString_Clear(encryptedData);    

        OpcUa_ByteString password;
        password.Length = utf8PasswordLen + serverNonce.Length + 4;
        password.Data = (OpcUa_Byte*)OpcUa_Alloc(password.Length);
        OpcUa_GotoErrorIfAllocFailed(password.Data);

        dataLength = password.Length - 4;
        OpcUa_MemCpy(password.Data, 4, &dataLength, 4);
        OpcUa_MemCpy(password.Data + 4, utf8PasswordLen, utf8Password, utf8PasswordLen);
        OpcUa_MemCpy(password.Data + utf8PasswordLen + 4, serverNonce.Data, serverNonce.Length);
        
        uStatus = OpcUa_CryptoProvider_Create(securityProfileUri, &cp);
        OpcUa_GotoErrorIfBad(uStatus);
        
        publicKey.Key.Length = 2048;
        publicKey.Key.Data = (OpcUa_Byte*)OpcUa_Alloc(publicKey.Key.Length);
        OpcUa_GotoErrorIfAllocFailed(password.Data);

        uStatus = cp.GetPublicKeyFromCert(&cp, &severCertificate, OpcUa_Null, &publicKey);
        OpcUa_GotoErrorIfBad(uStatus);
        
        encryptedData.Length = 2048;
        encryptedData.Data = (OpcUa_Byte*)Opc_Alloc(2048);
        OpcUa_GotoErrorIfAllocFailed(encryptedData.Data);

        uStatus = cp.AsymmetricEncrypt(&cp, password.Data, password.Length, &publicKey, encryptedData.Data,(OpcUa_UInt32*)&encryptedData.Length);
        OpcUa_GotoErrorIfBad(uStatus);

        OpcUa_CryptoProvider_Delete(&cp);

        OpcUa_ReturnStatusCode;
        OpcUa_BeginErrorHandling;

        OpcUa_CryptoProvider_Delete(&cp);
        OpcUa_ByteString_Clear(encryptedData);

        OpcUa_FinishErrorHandling;
    }

In this code i will suppose that the length of pulic key is 2048 bytes. It is right? How  can i to determine the length of public key data?
Another same question about the size of encrypted data. How can i  to determine the length of encrypted Data

sanny32, 2018-07-30:
I have forund a method to determine the size of public key and encrypted data. To call a method `GetPublicKeyFromCert` with a empty OpcUa_Key will return a size of public key. Then allocate memory with the given size and call `GetPublicKeyFromCert` again returns public key. The same way to determine the size of the encrypted data.

closed 2018-07-30 by sanny32

I-Yoo, 2020-04-09:
How did you implement the username/password connection?I hope you can reply me.

sanny32, 2020-04-09:
You must create OpcUa_UserNameIdentityToken, after that create extension object with OpcUa_EncodeableObject_CreateExtension, setup it with policyId, UserName, Password, and Encrypt it with server certificate and server nonce. Than you must put this token as parameter into OpcUa_ClientApi_ActivateSession

bernd-edlinger, 2020-04-09:
Just make sure your nonce is random 32 byte, regardless what policy you use.

I-Yoo, 2020-04-10:
After reading the UserIdentityToken parameter in Part 4-7.36 of the OPC UA specification, see that if the SecurityPolicy is None, then the password contains only the utf-8 encoded password. Is it not required to encrypt the password?

bernd-edlinger, 2020-04-10:
The security profile of the channel and the user token encryption are entirely different things.
One error that often happen, ist the server does not send a nonce, because the channel
is security-none, but the client want to send an encrypted token.  The client can not
fix that. If the client ignores the empty or all-zero token you have CVE-2019-19135.
That is as deadly as COVID-19 :smile:

https://opcfoundation.org/SecurityBulletins/OPC%20Foundation%20Security%20Bulletin%20CVE-2019-19135.pdf

I-Yoo, 2020-04-10:
@bernd-edlinger Do you mean that any SecurityPolicy requires encrypted passwords? My current problem is to implement Username/Password login to connect to Server.

bernd-edlinger, 2020-04-10:
Yes, unless you have other security measures like an encrypted ethernet tunnel (VPN),
or encypted WLAN (and you make sure that the WLAN traffic is not sent via
DSL because that is always unencrypted, as well as all internet traffic),
You would not connect your server to the internet, would you?

I-Yoo, 2020-04-10:
My server is connected to TCP locally. I want to implement the username and password login connection Demo. Could you give me some guidance?

bernd-edlinger, 2020-04-10:
Sorry, but my contract for maintaining the software was cancelled a long time ago.

bernd-edlinger, 2020-04-10:
But I am sure you got already a lot of guidance also from others in this thread.

So I'd say your chances are more than average, to implement your client
correctly.  Note there is a still maintained .NET server and client.
So make sure you test against them.  And there is a CTT test program,
by the OPC Foundation.  That cost some money but it helps to find common
programming errors.

I-Yoo, 2020-04-10:
@bernd-edlinger I'll consider your suggestion. Although my problem has not been solved, thank you for your patience.

bernd-edlinger, 2020-04-10:
one last suggestion: do not hurry too much, take your time, there is no need to panic.