OPCFoundation/UA-AnsiC-Legacy #121: Connect from OPC UA Client with user certificate.

sanny32, 2018-07-31:
I have write a OPC UA Client to connect to the OPC UA Server with certificate.
Exploring the .NET Stack i have found example `UserAuthentication client`. In the event handler `CertificateImpersonateBTN_Click` creates a` X509Certificate2` as

    // load the certficate.
    X509Certificate2 certificate = new X509Certificate2(
    CertificateTB.Text, 
    CertificatePasswordTB.Text, 
    X509KeyStorageFlags.MachineKeySet | X509KeyStorageFlags.Exportable);

My question is: what is the way to do the same with UA-AnsiC stack?

sanny32, 2018-08-01:
I have found a solution. From [CertificateGenerator ](https://github.com/OPCFoundation/Misc-Tools)project use this function 

    /**
     * @brief Loads a private key from a file.
     *
     * @return Status code; @see opcua_statuscodes.h
     */
    OpcUa_StatusCode OpcUa_Certificate_LoadPrivateKeyFromFile(
        OpcUa_StringA       a_sFilePath,  
        OpcUa_P_FileFormat  a_eFileFormat,
        const char*         a_sPassword,
        OpcUa_ByteString*   a_pCertificate,
        OpcUa_Key*          a_pPrivateKey);

closed 2018-08-01 by sanny32

I-Yoo, 2020-04-13:
Hello,I'd like to ask you about user name / password login and certificate login. Would you like to help me solve some problems?