OPCFoundation/UA-AnsiC-Legacy #152: How to configure clientNonce ?

425577456liu, 2020-04-14:
When I use the Publisher_CreateSession service,its parameter responseHeader.ServiceResult returns an error(OpcUa_BadNonceInvalid).
I initialized it like this :
	OpcUa_ByteString clientNonce;
	OpcUa_ByteString_Initialize(&clientNonce);
uStatus = OpcUa_ClientApi_CreateSession(
		a_pSession->Channel,
		&requestHeader,
		&clientDescription,
		&a_pSession->ApplicationUri,
		&a_pSession->EndpointUrl,
		&clientDescription.ApplicationName.Text,
		&clientNonce,
		&Publisher_g_ClientCertificate,
		requestedSessionTimeout,
		maxResponseMessageSize,
		&responseHeader,
		&a_pSession->SessionId,
		&a_pSession->AuthenticationToken,
		&a_pSession->RevisedSessionTimeout,
		&a_pSession->ServerNonce,
		&a_pSession->ServerCertificate,
		&noOfServerEndpoints,
		&pServerEndpoints,
		&noOfServerSoftwareCertificates,
		&pServerSoftwareCertificates,
		&serverSignature,
		&maxRequestMessageSize);
I am looking forward to your reply,thanks.

cacamille3, 2020-04-14:
You got `OpcUa_BadNonceInvalid `because you initialized your clientNonce but you need to set a valid Nonce value !

> clientNonce - A random number that should never be used in any other request. This number shall have a minimum length of 32 bytes. Profiles may increase the required length. The Server shall use this value to prove possession of its Application Instance Certificate in the response.

So you need basically to do something like

```
OpcUa_ByteString clientNonce;
OpcUa_ByteString_Initialize(&clientNonce); // <- Actually not needed anymore....

OpcUa_UInt32 nonceLength = 32;  //replace with your required nonce length
clientNonce.Length = nonceLength ; //replace with your required nonce length
clientNonce.Data =  OpcUa_Alloc(sizeof(OpcUa_Byte) * clientNonce.Length);
OpcUa_ReturnErrorIfAllocFailed(clientNonce.Data);
for(int i = 0; i < nonceLength; i++)
{
  clientNonce.Data[i] = (OpcUa_Byte)(rand() * GetTickCount()) % rand();
}
```

425577456liu, 2020-04-14:
Thank you very much,but responseHeader.ServiceResult returns an error(OpcUa_BadCertificateUriInvalid).
I did not re-create the certificate, I am using the certificate provided in the source code.
define UACLIENTNAME              "Opc.Ua.Publisher"
define UACLIENTURI               "urn:localhost:OPCFoundation:AMQPPublisher"
define UACLIENT_PRODUCTURI		  "http://opcfoundation.org/UA/Publisher/"	
I defined it this way in the Publisher_CreateSession service:
OpcUa_String_AttachReadOnly(&clientDescription.ApplicationName.Text, UACLIENTNAME);
OpcUa_String_AttachReadOnly(&clientDescription.ApplicationUri, UACLIENTURI);
OpcUa_String_AttachReadOnly(&clientDescription.ProductUri, UACLIENT_PRODUCTURI);
If my problem is here, Would you tell me how to solve it?
If I have to re-create the certificate, Would you tell me some ways?

Thank you again

cacamille3, 2020-04-14:
The OPC UA Server is checking that the ApplicationUri provided by the Client in its Client Description is matching the ApplicationUri set in the OPC UA Certificate Subject Alternative Name (SAN)

- The right way will be to create a new Certificate that will match your UACLIENTURI.
- A workaround will be to do the opposite and change your UACLIENTURI to match the value in your certificate.

425577456liu, 2020-04-15:
Thank you very much,I am sorry to bother you again ,I have changed the UACLIENTURI to match the value in the certificate.It no longer returns an error.
But in  OpcUa_ClientApi_ActivateSession,its parameter responseHeader.ServiceResult returns an error(OpcUa_BadSessionIdInvalid).It means the session id is not valid.
The parameter SessionId returned by OpcUa_ClientApi_CreateSession has data.
Is it possible that the problem appears in requestHeader.AuthenticationToken?

OpcUa_StatusCode Publisher_ActivateSession(Session* a_pSession)
{
	OpcUa_ReturnErrorIfArgumentNull(a_pSession);

	OpcUa_RequestHeader requestHeader;
	OpcUa_ResponseHeader responseHeader;
	OpcUa_SignatureData clientSignature;
	OpcUa_SignatureData identityTokenSignature;
	OpcUa_Int32 nResultCount = 0;
	OpcUa_StatusCode* pResults = NULL;
	OpcUa_Int32 nDiagnosticInfoCount = 0;
	OpcUa_DiagnosticInfo* pDiagnosticInfos = NULL;
	OpcUa_ByteString revisedNonce;

	OpcUa_InitializeStatus(OpcUa_Module_Client, "Publisher_ActivateSession");

	OpcUa_RequestHeader_Initialize(&requestHeader);
	OpcUa_ResponseHeader_Initialize(&responseHeader);
	OpcUa_SignatureData_Initialize(&clientSignature);
	OpcUa_SignatureData_Initialize(&identityTokenSignature);
	OpcUa_ByteString_Initialize(&revisedNonce);

	uStatus = OpcUa_NodeId_CopyTo(&a_pSession->AuthenticationToken, &requestHeader.AuthenticationToken);
	OpcUa_GotoErrorIfBad(uStatus);

	requestHeader.TimeoutHint = 60000;
	requestHeader.Timestamp = OpcUa_DateTime_UtcNow();
	uStatus = OpcUa_ClientApi_ActivateSession(
		a_pSession->Channel,
		&requestHeader,
		&clientSignature,
		0,
		NULL,
		0,
		NULL,
		&a_pSession->IdentityToken,
		&identityTokenSignature,
		&responseHeader,
		&revisedNonce,
		&nResultCount,
		&pResults,
		&nDiagnosticInfoCount,
		&pDiagnosticInfos);
}
Thank you again!

cacamille3, 2020-04-15:
Which OPC UA Server are you connecting to ? 
Are you connecting with the Security Policy **None** and **Anonymous** as Credentials ?
Are you sure the parameter ` const OpcUa_ExtensionObject*           a_pUserIdentityToken` is well initialized ?
You also need to increment the `requestHeader.RequestHandle` value for each Request

425577456liu, 2020-04-16:
The server I use for anonymous connection is “Prosys OPC  UA SimulationServer”,and I can't connect to Kepware too.
  Security Policy = OpcUa_SecurityPolicy_Basic128Rsa15
  messageSecurityMode= OpcUa_MessageSecurityMode_Sign
My a_pUserIdentityToken is initialized by the method provided by the demo,its code is as follows:
OpcUa_StatusCode Publisher_SelectAnonymousUserTokenPolicy(Session* a_pSession, OpcUa_EndpointDescription* a_pEndpoint)
 {
	OpcUa_ReturnErrorIfArgumentNull(a_pSession);
	OpcUa_ReturnErrorIfArgumentNull(a_pEndpoint);

	OpcUa_UserTokenPolicy* pAnonynmous = NULL; 

	OpcUa_UserTokenPolicy* pToken = NULL; // need to choose a user identity token.

	OpcUa_InitializeStatus(OpcUa_Module_Client, "Publisher_SelectAnonymousUserTokenPolicy");

	for (int ii = 0; ii < a_pEndpoint->NoOfUserIdentityTokens; ii++)
	{
		pToken = &a_pEndpoint->UserIdentityTokens[ii];

		if (pToken->TokenType == OpcUa_UserTokenType_Anonymous)
		{
			// We choose anonymous
			pAnonynmous = &a_pEndpoint->UserIdentityTokens[ii];
			break;
		}

		pToken = NULL;
	}

	if (pToken != NULL)
	{
		OpcUa_AnonymousIdentityToken* pBody = (OpcUa_AnonymousIdentityToken*)OpcUa_ExtensionObject_CreateFromType(&a_pSession->IdentityToken, &OpcUa_AnonymousIdentityToken_EncodeableType);
		OpcUa_GotoErrorIfNull(pBody, OpcUa_BadOutOfMemory);

		uStatus = OpcUa_UaString_CopyTo(&pAnonynmous->PolicyId, &pBody->PolicyId);
		OpcUa_GotoErrorIfBad(uStatus);

		pToken = pAnonynmous;

		a_pSession->IdentityTokenPolicy.TokenType = pToken->TokenType;
		uStatus = OpcUa_UaString_CopyTo(&pToken->PolicyId, &a_pSession->IdentityTokenPolicy.PolicyId);
		OpcUa_GotoErrorIfBad(uStatus);

		uStatus = OpcUa_UaString_CopyTo(&pToken->SecurityPolicyUri, &a_pSession->IdentityTokenPolicy.SecurityPolicyUri);
		OpcUa_GotoErrorIfBad(uStatus);

		uStatus = OpcUa_UaString_CopyTo(&pToken->IssuedTokenType, &a_pSession->IdentityTokenPolicy.IssuedTokenType);
		OpcUa_GotoErrorIfBad(uStatus);

		uStatus = OpcUa_UaString_CopyTo(&pToken->IssuerEndpointUrl, &a_pSession->IdentityTokenPolicy.IssuerEndpointUrl);
		OpcUa_GotoErrorIfBad(uStatus);
	}
	else
	{
		OpcUa_Trace(OPCUA_TRACE_LEVEL_WARNING, "Publisher_SelectAnonymousUserTokenPolicy: No anonymous user token policy supported by server endpoint.\n");
	}

	OpcUa_ReturnStatusCode;
	OpcUa_BeginErrorHandling;
	OpcUa_FinishErrorHandling;
 }

Now in OpcUa_ClientApi_ActivateSession,its parameter responseHeader.ServiceResult returns an error(OpcUa_BadApplicationSignatureInvalid).It means the signature generated with the client certificate is missing or invalid.
 In the previous code, clientSignature has no data,so I configure it through the following code:
```
        OpcUa_SignatureData             clientSignature;
	OpcUa_ByteString                 a_data;
	unsigned char a[] = "this is a data";
	a_data.Data = a;
	a_data.Length = sizeof(a);
	OpcUa_CryptoProvider    a_pProvider;
	OpcUa_StatusCode	 ret_OpcUa_CryptoProvider_Create=OpcUa_CryptoProvider_Create(
				(OpcUa_StringA)"http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15",
							&a_pProvider);
	OpcUa_ByteString*          a_pSignature = new OpcUa_ByteString;
	OpcUa_String* temp_al = new OpcUa_String;

	char ch_al[] = "http://www.w3.org/2000/09/xmldsig#rsa-sha1";//May be wrong here

	temp_al->strContent = ch_al;
	temp_al->uLength = sizeof(ch_al);

	temp_al->flags = 0;//May be wrong here

	OpcUa_StatusCode  ret_OpcUa_Crypto_AsymmetricSign = 
 OpcUa_Crypto_AsymmetricSign(&a_pProvider, a_data, Publisher_g_ClientPrivateKey, a_pSignature);
	clientSignature.Signature = *a_pSignature;
	clientSignature.Algorithm = *temp_al;
```
Now the parameter ret_OpcUa_Crypto_AsymmetricSign  returns an error(OpcUa_BadInvalidArgument),It means one or more arguments are invalid.
I have been debugging for a long time, I can't find the reason, please help me at your convenience,thank you.

cacamille3, 2020-04-16:
Use Security Policy None to debug your code.
Anyway the SecurityPolicy Basic128Rsa15 is deprecated and should not be use in production environment.
When your code is working with None and you want to use a Secure OPC UA Communication, you will have to take a look how to generate your own Client Certificate and also how to compute the ClientSignature passed in a ActivateSession.
Everything is well detailled in OPC UA Specification Part 4 - Services (https://opcfoundation.org/developer-tools/specifications-unified-architecture)

425577456liu, 2020-04-16:
I have tried Use Security Policy None to debug my code and it succeeded.So my next step will be how to create my own certificate.
Thank you very much.

closed 2020-06-03 by 425577456liu